redoakcanyon/HATPowerBoard #5: rocpmd won't stop

tgaryet, 2015-11-28:
Issuing:

sudo rocpmd stop

should stop the daemon.  Syslog has the following:

Nov 28 10:51:58 raspberrypi rocpmd: lockfile_path.c_str() = /var/run/rocpmd.pid
Nov 28 10:51:58 raspberrypi rocpmd: kill(atoi(pid.c_str()) = 0
Nov 28 10:51:58 raspberrypi rocpmd: rocpmd_status=3
Nov 28 10:51:58 raspberrypi rocpmd: pid = 548

vargur, 2015-11-28:
I'm not sure what you are trying to do, rocpmd stop does nothing since the daemon does not implement an option called stop.  On Wheezy rocpmd.sh stop should call the init script. If this stop option is a systemd requirement I can add it.

Mvh
Kristján Rúnarsson

tgaryet, 2015-11-28:
oh, never mind.  my misreading of the man page

closed 2015-11-28 by tgaryet