redoakcanyon/HATPowerBoard #8: rocpmd reports "Charging..." when run in command mode with no daemon alive

tgaryet, 2015-12-01:
If rocpmd --battery-level or --battery-level-raw  is issued a background daemon alive, the return is **Charging...** *even though there is no power applied*.  **Charging...** is returned for several seconds after the daemon is started.

pi@raspberrypi ~/HATPowerBoard $ sudo rocpmd --battery-level
Initializing GPIO...
Charging...
pi@raspberrypi ~/HATPowerBoard $ /etc/init.d/rocpmd start
-bash: /etc/init.d/rocpmd: No such file or directory
**pi@raspberrypi ~/HATPowerBoard $ sudo /etc/init.d/rocpmd.sh start**
Starting rocpmd...
Initializing GPIO...
pi@raspberrypi ~/HATPowerBoard $ sudo rocpmd --battery-level
Charging...
pi@raspberrypi ~/HATPowerBoard $ sudo rocpmd --battery-level
Charging...
pi@raspberrypi ~/HATPowerBoard $ sudo rocpmd --battery-level
Percentage power level: 24%

tgaryet, 2015-12-01:
System log reports:

Dec  1 12:47:14 raspberrypi rocpmd: Initializing GPIO...
Dec  1 12:47:14 raspberrypi rocpmd: connect() failed in function ud_client_send_command(): Connection refused
Dec  1 12:49:49 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Dec  1 12:50:49 2015 [try http://www.rsyslog.com/e/2007 ]
Dec  1 12:49:52 raspberrypi systemd[1]: Reloading.
Dec  1 12:50:08 raspberrypi rocpmd: Initializing GPIO...
Dec  1 12:50:09 raspberrypi rocpmd: connect() failed in function ud_client_send_command(): Connection refused
Dec  1 12:50:40 raspberrypi rocpmd: Initializing GPIO...
Dec  1 12:50:42 raspberrypi rocpmd: Starting rocpmd (pid: 12214)
Dec  1 12:50:52 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Dec  1 12:51:52 2015 [try http://www.rsyslog.com/e/2007 ]
Dec  1 13:00:41 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Tue Dec  1 13:01:41 2015 [try http://www.rsyslog.com/e/2007 ]

vargur, 2015-12-01:
I think I fixed this. Before the daemon starts it’s main loop it now does one battery level read and then waits a few seconds before doing another periodic battery level read. The interval is settable with the battery-level-read-interval field in the config file.

The problem is that it keeps dropping in and out on me, i.e sometimes the daemon reads the wall wart signal and returns -1 for charging but then it reverts to returning the battery level even though the wall wart is plugged in. I think I may have a loose connector somewhere. Can you take a look at this?

I put a couple of debugs into the daemon main loop, you should see something like the following in syslog:

ec  1 21:40:41 raspberrypi rocpmd: Initializing GPIO...
Dec  1 21:40:42 raspberrypi rocpmd: Starting rocpmd (pid: 6976)
Dec  1 21:40:43 raspberrypi dhcpcd[370]: wlan0: Router Advertisement from fe80::a6b1:e9ff:fe9c:ad17
Dec  1 21:40:44 raspberrypi rocpmd: battery_level_raw = 5
Dec  1 21:40:44 raspberrypi rocpmd: battery_level_percent = 100

Plug in wall wart...

Dec  1 21:41:12 raspberrypi rocpmd: battery_level_raw = -1
Dec  1 21:41:12 raspberrypi rocpmd: battery_level_percent = -1

Vive valeque,
Kristján Rúnarsson

tgaryet, 2015-12-01:
The strange part is that if one uses the raspi-gpio get command to look 
at the values on 20/26, then it reports the correct values depending on 
the actual
charger status.

This sounds like there is a contention on the actual pin - one thing is 
pulling it up, another is pulling it down, so that the voltage level is 
"mid" level, and it is a
coin toss which logic value get read.

I had never tested the battery-level function without the daemon running 
(never thought of it) so I don't know if the issue was introduced on the 
last commit or not.

I'm going to revert my install to before you made the hard coded. After 
that, I'm getting my voltmeter out.  Maybe I'll reverse these two .

Terry

vargur, 2015-12-01:
I just pushed my changes. I got side tracked measuring my connectors and wires and they all seem to be OK.

The strange thing is that the flutter is gone now.

Vive valeque,
Kristján Rúnarsson

vargur, 2015-12-01:
The problem was that the daemon was waiting 10 seconds before it read the battery level and updated the UD server, it now updates the UD server straight off the bat. I knew about this but did not fix it because the problem was gone by the time the computer had booted and the odds of some user figuring this out were pretty low and even if they did, it’s a configurable time interval. They can shorten it or just wait a few seconds. the battery does not drain so fats they need battery level updates at 1 second intervals.

Vive valeque,
Kristján Rúnarsson

tgaryet, 2015-12-01:
So 10 seconds after it starts, it (used) to take its first measurement.  
But right off the bat, it reported "Charging" when it really wasn't.

Is that the same problem?

If the daemon never daemonizes, it never lasts 10 seconds.  I have to 
admit I have no clue what this UD server is about.

Terry

tgaryet, 2015-12-01:
There is another problem me thinks.  This is from syslog with no daemon 
running in the background (ie, it is stopped)

Dec  1 16:10:05 raspberrypi rsyslogd-2007: action 'action 17' suspended, 
next retry is Tue Dec  1 16:10:35 2015 [try http://www.rsyslog.com/e/2007 ]
Dec  1 16:10:20 raspberrypi rocpmd: Initializing GPIO...
Dec  1 16:10:20 raspberrypi rocpmd: connect() failed in function 
ud_client_send_command(): Connection refused
Dec  1 16:10:28 raspberrypi rocpmd: Initializing GPIO...
Dec  1 16:10:28 raspberrypi rocpmd: connect() failed in function 
ud_client_send_command(): Connection refused
Dec  1 16:10:32 raspberrypi rocpmd: Initializing GPIO...
Dec  1 16:10:32 raspberrypi rocpmd: connect() failed in function 
ud_client_send_command(): Connection refused
Dec  1 16:10:39 raspberrypi rsyslogd-2007: action 'action 17' suspended, 
next retry is Tue Dec  1 16:11:09 2015 [try http://www.rsyslog.com/e/2007 ]
Dec  1 16:10:39 raspberrypi rocpmd: Initializing GPIO...
Dec  1 16:10:39 raspberrypi rocpmd: connect() failed in function 
ud_client_send_command(): Connection refused
(END)

I still get charging returned even though I'm not plugged in.

Terry

vargur, 2015-12-01:
Hi,
rocpmd used to:
Assume the board was charging.
Wait 10 seconds.
Make it’s first measurement.
...

Now rocpmd does this:
Makes it’s first measurement.
Wait 10 seconds.
Makes it’s second measurement.
...

The problem is that in steps 5 and after the first couple of times I tested the new way of doing this the daemon did not seem to be picking up on the fact that the wall wart was plugged in. It would return battery level -1 (charging) one or more times and then revert to reporting raw battery level 4 (100% even though the wall wart was plugged in and it should have been detecting the D2 signal from the board. This could be due to a funky connector or it could be due to GPIO problems? rocpmd? sysfs? pull up resistor?. I have no idea what.

I have never tested this feature, i.e. whether the thing consistently reports -1 when the wall wart is plugged in nor have I conducted extensive tests as to whether the battery level is correctly reported as the board slowly discharges. I just trusted the hardware.

Vive valeque,
Kristján Rúnarsson

> On 01 Dec 2015, at 22:02, Terry Garyet <notifications@github.com> wrote:
> 
> If the daemon never daemonizes, it never lasts 10 seconds. I have to

vargur, 2015-12-01:
Let’s solve one problem at a time. I thought this GPIO shit was fixed and now it is a steaming mess again. I’m getting really depressed over here, this GPIO junk is evolving into a never ending story.

On what OS are you doing this and on what board revision? 

Vive valeque,
Kristján Rúnarsson

> On 01 Dec 2015, at 22:12, Terry Garyet <notifications@github.com> wrote:
> 
> Dec 1 16:10:20 raspberrypi rocpmd: Initializing GPIO...

tgaryet, 2015-12-01:
This is behaving exactly the same, even with the 10 second delay before 
I get real battery levels.

I don't think I got your latest commit even though I did a pull

tgaryet, 2015-12-01:
I wouldn't say steaming quite yet.  This is a weird corner case I think.

I am using jessie with a+.

The daemon is stopped and I type
sudo rocpmd --battery-level and it returns "Charging" even if the AC is 
unplugged.

Having said this, when you said you fixed it, I pulled and re made, but 
I don't think I got your update.  git log doesn't have your commit in it 
that I can see.  Here are the last commits and I just did a pull again:

pi@raspberrypi ~/HATPowerBoard $ git log
commit d60653f244fe5a11b293541ddaa1048750feb774
Author: Terry Garyet <terry@redoakcanyon.com>
Date:   Tue Dec 1 15:58:03 2015 -0600

     initial commit of man page for device-tree

commit 936e4abfa031b2b048f1a8e28a95829dae9de710
Author: Terry Garyet <terry@redoakcanyon.com>
Date:   Tue Dec 1 14:44:51 2015 -0600

     Fixed a few additonal typos

commit b1a591b9ff414d4e91f31b37b57d4a7e5b0184a3
Author: Terry Garyet <terry@redoakcanyon.com>
Date:   Tue Dec 1 14:28:54 2015 -0600

     Updated README to reflect actual required install process and
     added other hardare related info

I obviously do not have your last fix.  (breath)
Terry

vargur, 2015-12-01:
Once again git silently failed to push my changes but I think I fixed it now, at least my changes are on the github web site now. I am really beginning to hate git.

Vive valeque,
Kristján Rúnarsson

tgaryet, 2015-12-01:
now i got em.

When you push, if I had pushed since you pulled, then you have to pull 
first.  There is so much text in the push message it is hard to see that 
it failed.

git is quite the complex beast.

Terry

vargur, 2015-12-01:
OK, I did an experiment:

Plugged the wall wart in and made sure that the daemon was reporting a battery level of -1.
Waited until it started reporting battery level 100% (raw=4) again.
Power cycled the wall wart with the switch on the wall connector.
The daemon instantly started reporting the wall wart was plugged in.

Vive valeque,
Kristján Rúnarsson

vargur, 2015-12-01:
Has now reverted to returning the battery level, wall wart still plugged in.

Vive valeque,
Kristján Rúnarsson

tgaryet, 2015-12-01:
Now, with your latest version it still behaves the same way.  I don't 
honestly remember which
one is d1 and d2 but they are and should be tracking each other at the 
very moment.  As I
peek using sysfs:

No charger plugged in:
pi@raspberrypi /sys/class/gpio $ cat gpio20/value gpio26/value
1
1
With charger just plugged in:
pi@raspberrypi /sys/class/gpio $ cat gpio20/value gpio26/value
0
0
pi@raspberrypi /sys/class/gpio $

pi@raspberrypi /sys/class/gpio $ sudo rocpmd --battery-level-raw
Initializing GPIO...
Charging...

with our without the charger plugged in.

Then start and get

pi@raspberrypi /sys/class/gpio $ sudo /etc/init.d/rocpmd.sh start
Starting rocpmd...
Initializing GPIO...
pi@raspberrypi /sys/class/gpio $ sudo rocpmd --battery-level-raw
Raw power level:        5
pi@raspberrypi /sys/class/gpio $

and when I plug in:

Got charging after a very short delay.  This is actually correct, as it 
takes a second for the charging chip to decide that the battery is not 
shorted before
it really starts.  This is < 1 second, not the seconds worth of delay to 
get a real reading after a report of charging that I reported before.

Here is the syslog starting at a point before the daemon was started and 
ending when I plugged in the AC:

Dec  1 16:26:15 raspberrypi rocpmd: connect() failed in function 
ud_client_send_command(): Connection refused
Dec  1 16:26:22 raspberrypi rocpmd: Initializing GPIO...
Dec  1 16:26:22 raspberrypi rocpmd: connect() failed in function 
ud_client_send_command(): Connection refused
Dec  1 16:26:27 raspberrypi rocpmd: Initializing GPIO...
Dec  1 16:26:27 raspberrypi rocpmd: connect() failed in function 
ud_client_send_command(): Connection refused
Dec  1 16:30:52 raspberrypi rsyslogd-2007: action 'action 17' suspended, 
next retry is Tue Dec  1 16:31:22 2015 [try http://www.rsyslog.com/e/2007 ]
Dec  1 16:30:52 raspberrypi rocpmd: Initializing GPIO...
Dec  1 16:30:52 raspberrypi rocpmd: connect() failed in function 
ud_client_send_command(): Connection refused
Dec  1 16:31:01 raspberrypi rocpmd: Initializing GPIO...
Dec  1 16:31:01 raspberrypi rocpmd: connect() failed in function 
ud_client_send_command(): Connection refused
Dec  1 16:31:05 raspberrypi rocpmd: Initializing GPIO...
Dec  1 16:31:05 raspberrypi rocpmd: connect() failed in function 
ud_client_send_command(): Connection refused
Dec  1 16:32:49 raspberrypi rsyslogd-2007: action 'action 17' suspended, 
next retry is Tue Dec  1 16:33:19 2015 [try http://www.rsyslog.com/e/2007 ]
Dec  1 16:32:50 raspberrypi rocpmd: Initializing GPIO...
Dec  1 16:32:50 raspberrypi rocpmd: Starting rocpmd (pid: 11559)
Dec  1 16:32:52 raspberrypi rocpmd: battery_level_raw = 5
Dec  1 16:32:52 raspberrypi rocpmd: battery_level_percent = 100
Dec  1 16:33:02 raspberrypi rocpmd: battery_level_raw = 5
Dec  1 16:33:02 raspberrypi rocpmd: battery_level_percent = 100
Dec  1 16:33:12 raspberrypi rocpmd: battery_level_raw = 5
Dec  1 16:33:12 raspberrypi rocpmd: battery_level_percent = 100
Dec  1 16:33:23 raspberrypi rocpmd: battery_level_raw = 5
Dec  1 16:33:23 raspberrypi rsyslogd-2007: action 'action 17' suspended, 
next retry is Tue Dec  1 16:33:53 2015 [try http://www.rsyslog.com/e/2007 ]
Dec  1 16:33:23 raspberrypi rocpmd: battery_level_percent = 100
Dec  1 16:33:33 raspberrypi rocpmd: battery_level_raw = 5
Dec  1 16:33:33 raspberrypi rocpmd: battery_level_percent = 100
Dec  1 16:33:43 raspberrypi rocpmd: battery_level_raw = 5
Dec  1 16:33:43 raspberrypi rocpmd: battery_level_percent = 100
Dec  1 16:33:53 raspberrypi rocpmd: battery_level_raw = 5
Dec  1 16:33:53 raspberrypi rsyslogd-2007: action 'action 17' suspended, 
next retry is Tue Dec  1 16:34:53 2015 [try http://www.rsyslog.com/e/2007 ]
Dec  1 16:33:53 raspberrypi rocpmd: battery_level_percent = 100
Dec  1 16:34:00 raspberrypi rocpmd: battery_level_raw = -1
Dec  1 16:34:00 raspberrypi rocpmd: battery_level_percent = -1
Dec  1 16:34:10 raspberrypi rocpmd: battery_level_raw = -1
Dec  1 16:34:10 raspberrypi rocpmd: battery_level_percent = -1
:

SUMMARY:  The daemon is working properly when there is a daemon started 
and running.  It only returns charging when the daemon is stopped.

Terry

vargur, 2015-12-01:
I just pushed a fix. After that fix I am getting the following behaviour:

When the daemon is running the battery level it reads in now being reported when running a non daemon instance with --battery-level.
When the daemon is NOT running the battery level it reads in now being reported when running a non daemon instance with --battery-level.
The battery level being read is wrong some of the time. 
Even with the wall wart plugged in it reports battery level 100% (raw=4).
If I power cycle the wall wart it starts reporting that the wall wart is installed, i.e. raw power level -1.
After about a minute it loses track and starts reporting battery level 100% (raw=4) even though the wall wart is still plugged in.

Vive valeque,
Kristján Rúnarsson

> On 01 Dec 2015, at 22:39, Terry Garyet <notifications@github.com> wrote:
> 
> SUMMARY: The daemon is working properly when there is a daemon started 
> and running. It only returns charging when the daemon is stopped.

vargur, 2015-12-01:
There are no longer any connect errors on my side.

Vive valeque,
Kristján Rúnarsson

> On 01 Dec 2015, at 22:49, Kristján Rúnarsson <kristjanr@me.com> wrote:
> 
> I just pushed a fix. After that fix I am getting the following behaviour:
> 
> When the daemon is running the battery level it reads in now being reported when running a non daemon instance with --battery-level.
> When the daemon is NOT running the battery level it reads in now being reported when running a non daemon instance with --battery-level.
> The battery level being read is wrong some of the time. 
> Even with the wall wart plugged in it reports battery level 100% (raw=4).
> If I power cycle the wall wart it starts reporting that the wall wart is installed, i.e. raw power level -1.
> After about a minute it loses track and starts reporting battery level 100% (raw=4) even though the wall wart is still plugged in.
> 
> 
> Vive valeque,
> Kristján Rúnarsson
> 
>> On 01 Dec 2015, at 22:39, Terry Garyet <notifications@github.com <mailto:notifications@github.com>> wrote:
>> 
>> SUMMARY: The daemon is working properly when there is a daemon started 
>> and running. It only returns charging when the daemon is stopped.
>

tgaryet, 2015-12-01:
What is the battery level when you unplug the charger?  Is your battery 
all charged up?

Terry

vargur, 2015-12-01:
Battery level with wall wart unplugged in 4.09 volts.

Vive valeque,
Kristján Rúnarsson

tgaryet, 2015-12-01:
Are you monitoring the charge led or the power led.  The charge led will 
go off when the battery is fully charged.

When you initially plug in the power, even with a fully charged battery, 
it will "top it off" .  It actually starts the charge
cycle, figures and stops when the charge current drops below some threshold.

I'm buiding it right now.. Give me a few to check it out.

Terry

tgaryet, 2015-12-01:
No i meant the % level.

Terry

vargur, 2015-12-01:
it is 100% (raw=4).

Vive valeque,
Kristján Rúnarsson

> On 01 Dec 2015, at 22:56, Terry Garyet <notifications@github.com> wrote:
> 
> up

vargur, 2015-12-01:
Ahaaaa! ...that could be the problem. Which is which?

Vive valeque,
Kristján Rúnarsson

vargur, 2015-12-01:
I’m reading D2 in the battery read routine before I do the actual battery read.

Vive valeque,
Kristján Rúnarsson

tgaryet, 2015-12-01:
exactly.  so do this:
pi@raspberrypi ~/HATPowerBoard $ cat /sys/class/gpio/gpio20/value 
/sys/class/gpio/gpio26/value

My system/battery are in excatly the same state:
pi@raspberrypi ~/HATPowerBoard $ cat /sys/class/gpio/gpio20/value 
/sys/class/gpio/gpio26/value
1
0
pi@raspberrypi ~/HATPowerBoard $

Plugged in but not charging.  Meanwhile:

pi@raspberrypi ~/HATPowerBoard $ sudo rocpmd --battery-level
Initializing GPIO...
Percentage power level: 100%
pi@raspberrypi ~/HATPowerBoard $

Is absolutely correct.  (That was without daemon running).

pi@raspberrypi ~/HATPowerBoard $ sudo /etc/init.d/rocpmd.sh start
Starting rocpmd...
Initializing GPIO...
pi@raspberrypi ~/HATPowerBoard $ sudo rocpmd --battery-level
Percentage power level: 100%
pi@raspberrypi ~/HATPowerBoard $

It is working just fine now.  Is that delay when no daemon running 
necessary?

Terry

vargur, 2015-12-01:
I don’t understand?

Vive valeque,
Kristján Rúnarsson

tgaryet, 2015-12-01:
Let me check the schematic for a minute.  You want to be reading the 
charge LED, not the power LED.  I think you have it correct!!

vargur, 2015-12-01:
Well that is disappointing if D2 is the correct pin to read that can’t be the problem.

For what it is worth:
D1 is mapped to BCM GPIO26 in the daemon.
D2 is mapped to BCM GPIO20 in the daemon.

Vive valeque,
Kristján Rúnarsson

tgaryet, 2015-12-01:
> 1 (20 == D1_B -- not charging)
> 0 (26 == D2_B -- powered)

D1 low means the battery is charging
D2 low means power is applied

The battery only charges until it is fully charged.  There isn't a 
trickle charge for LiOn batteries.

Now, the error is in the config file.  It has:
     - bcm-pin: 26
       role: D1_B
     - bcm-pin: 20
       role: D2_B

When in fact, D1 is connected to pin 20, and D2 is connected to 26.

This is totally a question of preference.  Do you want to know when it 
is powered or charged or both?

Terry

tgaryet, 2015-12-01:
But you still fixed it!

I'm thinking it is time to freeze this code.  I've made lots of progress 
on the documentation today, added my roc-device-tree man page, added 
sections to the readme to describe the rest of the hardware on the board.

Terry

vargur, 2015-12-01:
God I’m an idiot!!!!

So let’s get this straight once and for all :-)
D1 is physically connected to BCM20. It is high when battery is charging, low when battery is not charging.
D2 is physically connected to BCM26. It is high when wall wart is connected, low when wall wart is not connected.
This mistake must have happened when I purged the WiringPi crap out of the code.

Vive valeque,
Kristján Rúnarsson

tgaryet, 2015-12-01:
These are active LOW signals:

D1*_B* is physically connected to BCM20. It is *LOW *when battery is 
charging, *HIGH *when battery is not charging.
D2*_B* is physically connected to BCM26. It is *LOW ***when wall wart is 
connected, *HIGH *when wall wart is not connected.

If you are talking about variable in your code, you are reading the 
logical value...you invert the 0 or 1 that you read from the gpio pin:

int gpio_read_d2(config *conf)
{
     int pin_no_d2 = conf->get_gpio_by_role(CFG_ROLE_D2_B).bcm_pin;

     int status = GPIO_SUCCESS;
     status = digitalRead (pin_no_d2);
     if(status == GPIO_ERROR) { return status; }

*    return !status;*
}

technically that should be:
status_b = digitalRead(pin_no_d2)
..
return !status_b

Terry

vargur, 2015-12-01:
I inverted the logical value despite the active low thing because I’m used to 1 being true and 0 being false. It’s never the other way around In my world and it would have been unnecessarily confusing have some variables where 0 is ‘true’ and others where 1 is ‘true’. I’m used to the content of IF statements executing when x>0 and not when x==0.

You are right about the names, I’ll create a task to change the names in the gpio.cpp.

Vive valeque,
Kristján Rúnarsson

vargur, 2015-12-01:
There is still a bunch of cleaning up to do. So lets give it until Friday, assuming nothing else goes tits up.

Because you asked in another mail how I want the mapping to be:

I want D1 which is physically connected to BCM20 to be mapped as D1 in the daemon.
I want D2 which is physically connected to BCM26 to be mapped as D2 in the daemon.

The reason is that I want to use D2 to check whether the wall wart is charging.

To this end I have changed the config file as follows:

GPIOS:
    - bcm-pin: 6
      role: CS_B
    - bcm-pin: 12
      role: OFF
    - bcm-pin: 13
      role: PGOOD_B
    - bcm-pin: 16
      role: UD_B
    - bcm-pin: 5
      role: REQ_OFF_B
    - bcm-pin: 20
      role: D1_B
    - bcm-pin: 26
      role: D2_B

Vive valeque,
Kristján Rúnarsson

closed 2015-12-06 by vargur