redoakcanyon/HATPowerBoard #14: Not building on OctoPi image 0.13 (Jessie Lite)

WZ9V, 2016-07-23:
I followed the instructions and when I get to the make step I get the following:

cd src; make; cd -
make[1]: Entering directory '/home/pi/HATPowerBoard/src'
g++ -c -o obj/tdaemon.o tdaemon.cpp -g -std=c++11 -Wall -I../include
g++ -c -o obj/command-line.o command-line.cpp -g -std=c++11 -Wall -I../include
make[1]: *** No rule to make target 'config.cpp', needed by 'obj/config.o'.  Stop.
make[1]: Leaving directory '/home/pi/HATPowerBoard/src'
/home/pi/HATPowerBoard

WZ9V, 2016-07-23:
Looking closer it looks like you checked a link is config.cpp instead of the actual source file when it was updated 2 months ago.

tgaryet, 2016-07-23:
That should be fixed. If you

git status

Does it say you are behind one commit?

Try a

git fetch

To update your repository.

Let me know if you still have problems.

Terry

WZ9V, 2016-07-24:
Still gives me an error

pi@octopi:~/HATPowerBoard $ git status
On branch master
Your branch is up-to-date with 'origin/master'.
nothing to commit, working directory clean
pi@octopi:~/HATPowerBoard $ git fetch
pi@octopi:~/HATPowerBoard $ make
cd src; make; cd -
make[1]: Entering directory '/home/pi/HATPowerBoard/src'
g++ -c -o obj/tdaemon.o tdaemon.cpp -g -std=c++11 -Wall -I../include
g++ -c -o obj/command-line.o command-line.cpp -g -std=c++11 -Wall -I../include
make[1]: *** No rule to make target 'config.cpp', needed by 'obj/config.o'.  Stop.
make[1]: Leaving directory '/home/pi/HATPowerBoard/src'
/home/pi/HATPowerBoard

tgaryet, 2016-07-24:
Hi,

It looks like there is a second bad link in ../src.

Please fall back to a checkin that doesn't have this problem.

git checkout 1ca2a347df

then make, and then make install as per the instructions.

Sorry you are having problems building this.

Terry

WZ9V, 2016-07-24:
That one completed make and make install.  I'll check on operation tomorrow but so far it looks good.

closed 2016-07-24 by WZ9V

tgaryet, 2016-07-24:
Hi,

I've updated github so the changes associated with the broken links are now on a branch.  The master is effectively backed up to the version you checked out.

If you were to
Git pull
Git checkout master ( the default)

You should be able to build successfully.

Terry

Closed #14<https://github.com/redoakcanyon/HATPowerBoard/issues/14>.

-